ArduCAM/Arducam_tof_camera #16: Interlacing issue

luveti, 2022-10-26:
I followed the instructions here: https://www.arducam.com/docs/cameras-for-raspberry-pi/tof-camera-for-raspberry-pi/getting-started-with-arducam-tof-camera/

The output from the example programs shows the following:
[image]

dennis-ard, 2022-10-27:
Sorry to trouble you, The SDK version can be provided by the following command:
```Shel
dpkg -l | grep arducam
```
l

Andrew-rw, 2022-10-27:
I have exactly the same issue. Below are the output of `dpkg` and `uname`:
```
robot@raspberrypi:~ $ dpkg -l|grep arducam
ii  arducam-config-parser-dev              0.1.1                            armhf        Arducam config parser SDK
ii  arducam-tof-sdk-dev                    0.0.6                            armhf        Driver development kit for arducam tof camera
ii  arducam-usb-sdk-dev                    2.0.0                            armhf        Arducam USB SDK
robot@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
robot@raspberrypi:~ $ 

```

luveti, 2022-10-27:
`dpkg -l | grep arducam`
```
ii  arducam-config-parser-dev    0.1.1            arm64        Arducam config parser SDK
ii  arducam-tof-sdk-dev          0.0.6            arm64        Driver development kit for arducam tof camera
ii  arducam-usb-sdk-dev          2.0.0            arm64        Arducam USB SDK
```

`uname -a`
```
Linux raspberrypi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
```

dennis-ard, 2022-10-28:
We have tested and fixed this issue, please update the SDK with the following command
version to 0.0.7, and recompile the code.
```Shell
sudo apt update
sudo apt install arducam-tof-sdk-dev
```

Andrew-rw, 2022-10-28:
@dennis-ard, arducam-tof-sdk-dev v0.0.7 has fixed the issue! Thank you!
BTW, the 0.0.7 version is still not available in the repository, but I was able to download it from your pull-request (https://github.com/ArduCAM/arducam_ppa/pull/6)

luveti, 2022-10-28:
This has fixed the issue for me as well! Thank you for the quick fix! :)

closed 2022-10-28 by luveti