ArduCAM/Arducam_tof_camera #18: rgb + d in raspberry

monschine, 2022-11-13:
Hi, 
is it possible to connect to cameras to a raspberry pi 3 model b ?
I have seen there are some multi cam adapters, but as far as I can read they need to be same resolution, and because of the low resolution of the tof camera would that be still possible ?

Thanks

dennis-ard, 2022-11-16:
Hi,

monschine, 2022-11-16:
Thanks for the answer, I think it would be very valuable for the community if you could perform some tests and confirm that such configuration is working on raspberry pi with the multicamera adpaters.
if there is some computation limitation on my case would be acceptable to have this setup working from raspberry pi 4 and above...
Kind regards

dennis-ard, 2022-11-16:
Thank you for your proposal, we will include it as a supported function. We may not be able to reply to you in the near future, but we will support this function in the future.
Kind regards

vlsd, 2022-11-18:
FWIW we've tried it with an Arducam RGB camera through USB and were not able to get them both working at the same time. We didn't spend a lot of time trying though, as the errors we were getting were all over the place (kept changing every reboot).

v1ster, 2022-11-21:
If you still want to try, we will provide as much support as possible.

dennis-ard, 2022-11-21:
If you still want to try, we will provide as much support as possible.

dennis-ard, 2022-11-21:
In view of the complexity of this issue, please contact support@arducam.com, and we will assign someone to contact you.

monschine, 2022-11-22:
Thanks, 
I already contacted them. To be true I can not wait to start testing the system !
Just as extra bonus, I want to mention that the final goal for the system I thinking about, is to use 2 tof cameras + 2 rgb cameras. Let's see if the multicam adapter for 4 cameras could handle that

dennis-ard, 2022-11-23:
that's awesome!

vlsd, 2022-11-23:
> Let's see if the multicam adapter for 4 cameras could handle that

Interested in your results here as well, as we'd like to try it as well in the near future.

dennis-ard, 2022-11-24:
Oh, and I also asked the engineer about the use of the multicam adapter. If you use 4 cameras at the same time, you need the same model. If you use different models of cameras, they cannot be started at the same time to obtain frame data. If you want to connect two different models of cameras, you may need two camera ports to use the multicam adapter.

dennis-ard, 2022-12-02:
> FWIW we've tried it with an Arducam RGB camera through USB and were not able to get them both working at the same time. We didn't spend a lot of time trying though, as the errors we were getting were all over the place (kept changing every reboot).

hi,
We found that usb camera and tof can be used at the same time, but this needs to specify the video path of tof in the code, for example :
```C
int video_index = 1;
tof.init(ArduCam::CSI, ArduCam::DEPTH_TYPE, video_index );
```

monschine, 2022-12-19:
Thanks for the info !
So could a multiadpater be used to link 2 tof cameras as they are the same sensor ?

dennis-ard, 2023-01-11:
Thank you for your responding. 
At present, we have verified the use of multiple TOF cameras via USB. If you want us to verify the use of TOF cameras with multiadpater, you can contact support@arducam.com and give them feedback on your needs.