ArduCAM/Arducam_tof_camera #24: Can not install python ArduCamDepthCamera package on a jetson

adriaciurana, 2022-11-22:
Hello,
I'm trying to install the python package. Also I already installed the C++ library and works well.
But at the moment that I try to install ArduCamDepthCamera package, I obtain the following error:
`
Collecting ArduCamDepthCamera
  Could not find a version that satisfies the requirement ArduCamDepthCamera (from versions: )
No matching distribution found for ArduCamDepthCamera
`
Maybe is not available on the jetson?

Thanks, Adrià

adriaciurana, 2022-11-23:
> hi, We just tested that it can be installed, but pip needs to be updated to the latest version.

Oh true! Thanks.

One last question, in the C++ version I see that exists two methods setMode and setControl can be used. I don't see the possibility of having access using the python API.

On the other hand setControl when mode=RANGE, how is it different from setMode?

I want to use the 2m mode on python.

Thanks, Adria

dennis-ard, 2022-11-25:
hi,

dennis-ard, 2022-11-28:
It is my pleasure to serve you,
Better to reopen a question for a different problem.
About imx219 can not be detected, you can refer to the [website configuration jetson-io](https://www.arducam.com/docs/camera-for-jetson-nano/jetson-nano-xavier-nx-configuring-the-csi-connector-w-jetson-io/) .
Configure jetson-io as camera IMX219-A and Arducam-B, A and B correspond to camera interface 0 and 1.