ArduCAM/Arducam_tof_camera #28: Unable to manual set exposure value via setControl method

chinkit-ffc, 2022-12-20:
Hi, I am trying to adjust the FPS of ToF camera by explicitly setting the exposure parameter to higher value. The code can be compiled successfully and it runs fine, but the FPS still remains at the default value of approximately 30... 

I check the status code after calling tof.setControl function, it always return -1(Error). Changing the sequence of the function call to earlier line right after tof variable declaration also didn't help. I attach the code snippet below for your inspection, please let me know if you spot any mistake that is causing the issue. Or is there other function calls to set the exposure parameter??

I modify the preview_depth.cpp file provided in the example folder as follows: 
```
// in main function 
ArduCam::ArduCamTOFCamera tof;
ArduCam::FrameBuffer *frame;
if (tof.init(ArduCam::CSI,ArduCam::DEPTH_TYPE)){
    std::cerr<<"initialization failed"<<std::endl;
    exit(-1);
}

if (tof.start()){
    std::cerr<<"Failed to start camera"<<std::endl;
    exit(-1);
}

//  Modify the range also to modify the MAX_DISTANCE
tof.setControl(ArduCam::RANGE,MAX_DISTANCE);
int res = tof.setControl(ArduCam::EXPOSURE,65523);
std::cout << "Set exposure status: " << res << std::endl;

ArduCam::CameraInfo tofFormat = tof.getCameraInfo();
    
//The rest of the code is same as what originally written in the cpp file
....
```

Thanks in advance.

dennis-ard, 2023-01-11:
At present, the frame rate of the TOF camera is actually 120 frames, and the depth data is obtained by calculating the phase deviation through the data of 4 consecutive pictures. Therefore, the frame rate of the camera may not be higher.
Exposure is a reserved interface for debugging, so it cannot be used when it is officially released. If you still want to adjust the exposure, you can contact support@arducam.com to give feedback on your needs.

Undertaker7533967, 2023-02-21:
Hi ~~Dennis:

we also have arducam problem and sent it to support@arducam.com,  but only discobot always response me , how should I do ??

all the best~~