ArduCAM/Arducam_tof_camera #31: module 'ArducamDepthCamera' has no attribute 'ArducamCamera'

Coimbra1984, 2023-02-09:
Hi,

I installed the python dependencies (I work on a raspberry 4) as described in the README, however I get the following error message:

```
Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ArducamDepthCamera as ac
>>> cam = ac.ArducamCamera()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'ArducamDepthCamera' has no attribute 'ArducamCamera'

```

dennis-ard, 2023-02-22:
Hello,
Can you provide the SDK version you use? If the version of the SDK is less than 0.0.10, this may be the cause of the problem.
Please use the shell command to provide me with information:
```Shell
  pip3 list | Grep arducamdepthcamera
```
If the SDK version is less than 0.0.10, please try it again after updating.

Lunacyworks, 2023-07-24:
I am using 0.0.11 and get the following error

Traceback (most recent call last):
  File "/home/rpi4/Arducam_tof_camera/demo.py", line 51, in <module>
    if cam.init(ac.TOFConnect.CSI,ac.TOFOutput.DEPTH,0) != 0 :
AttributeError: 'ArducamDepthCamera.ArducamDepthCamera.ArducamCamer' object has no attribute 'init'