ArduCAM/Arducam_tof_camera #32: IM316 ToF Camera testing

Undertaker7533967, 2023-02-14:
Hi~ we have tested Arducam IMX316 ToF camera module on Raspberry Pi 4 which are confused on:

There are a big different distance results with three ROI on same object(smooth plane) as attached pictures:

1st ROI of left hand side , the distance is around 0.8
[image]

2nd of  Right hand side, the distance is around nan
![第二點3 32](https://user-images.githubusercontent.com/98312746/218611120-4cd

3rd of in the center , the distance is around 3.32
[image]
4ff58-56f8-49e1-abe7-f5fd5b3a59b5.jpg)

Are any Python API like C++ API and C API ?? when does it created or released ??

Are any Python example include (capture_raw.py) and (preview_depth.py) with more detail SDK commend and parameters explanation ??

all the best~~

dennis-ard, 2023-03-02:
Hi~
1. After our testing, we fixed the bug in this python example, thank you for your issue.
it was fixed in this PR. #34 
2. The document will be updated in the near future. The python api can refer to the use of the c api. They are consistent.
Best regard

Undertaker7533967, 2023-03-02:
Hallo~~Dennis:

1. we have got the upgrade python example and after testing it, the distance shown looks more accuracy than before.
2. We take reference C++API: for example int setExposure (int value) to cam.setExposure(100), we found there is no Python object   setExposure , so we don't know how to define Python API from C++ or C. as below picture:
[image]

Could you please provide us all Python like cam.setControl 

All the best~~