ArduCAM/Arducam_tof_camera #33: IMX316 drivers&libs for Ubuntu ARM64

andrew-bushuev, 2023-02-20:
The easiest way to use ROS2 is to rely on precompiled ROS2,
which is available for Ubuntu ARM64 (per the manual below)
https://docs.ros.org/en/humble/How-To-Guides/Installing-on-Raspberry-Pi.html

The question: are there any kernel modules/camera libraries for Ubuntu 22.04.1 LTS?
Will the kernel module and camera library source code be available publically?

dennis-ard, 2023-02-22:
The kernel modules are open source, and the code has been pushed to the Raspberry Pi warehouse as the [arducam-pivariety file](https://github.com/raspberrypi/linux/tree/rpi-5.15.y/drivers/media/i2c). The camera libraries are not open source for the time being, but we provide deb packages on Ubuntu 22.04.1 LTS.

andrew-bushuev, 2023-02-23:
> The kernel modules are open source, and the code has been pushed to the Raspberry Pi warehouse as the [arducam-pivariety [...]

Dennis, per the https://docs.arducam.com/Raspberry-Pi-Camera/Pivariety-Camera/Introduce/ I see that 64-bit Ubuntu Desktop 22.04 LTS no *not* supported. 
Also, per the file https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/media/i2c/arducam-pivariety.c I see no any imx316 mentions. 

Are there any updates I'm missing?
Thank you in advance.

dennis-ard, 2023-02-23:
pivariety is a TOF camera solution we provide. Ubuntu on Raspberry Pi is not supported yet. Maybe you can choose Jetson platform to run Arducam TOF Camera