ArduCAM/Arducam_tof_camera #35: IMX316 C program issue

Undertaker7533967, 2023-03-02:
We have one more question:

1. we have compiled the test.c for IMX316 Tof arducam
but there are no ArducamDepthCamera.h and it couldn't run the test.c
so please provide us ArducamDepthCamera.h to testing  IMX316 Tof arducam
2. how to do calibration of IMX316 Tof arducam ?? could you please release the calibration rule guide??

all the best~~

dennis-ard, 2023-03-02:
Hi~
1. ArducamDepthCamera.h is provided in the deb installation package of your installed arducam_tof_sdk_dev, please confirm whether you have installed arducam_tof_sdk_dev, you can use the following command to query:
```Shell
dpkg -l | grep arducam-tof-sdk-dev
ii  arducam-tof-sdk-dev                    0.0.10                           armhf        Driver development kit for arducam tof camera
```
If installed, please confirm whether `include_directories(/usr/include/c)` is included in CMakeLists.txt.
2.  TOF cameras use phase difference to calculate distance. Please refer to the relevant calibration documents for calibration. A calibration guide may be available later, but this may take longer.
Best regard

Undertaker7533967, 2023-03-02:
Dear Dennis:

thank you for your recommand methods, we will try to install cmake compile plateform to set up the ArducamDepthCamera.h 

all the best~~