ArduCAM/Arducam_tof_camera #36: Building Preview_pointcloud issue

Sv3ns0n, 2023-03-09:
Hello,

I have an issue with building the pointcloud.cpp demo. I followed the steps but it wont finish building.
Is there any solution to this problem?

dennis-ard, 2023-03-13:
Hi~,
Can you provide the command line output information when installing dependencies and compiling the startup program, as well as the running system and hardware model.
Best Regard

Sv3ns0n, 2023-03-14:
Hi, 
[image]
[image]

installing dependencies and compiling works, but apparently the code does not work:
[image]

Hardware model is Raspberry Pi Model 3A+ with Raspberry Pi OS (32-bit, Debian Bullseye)

dennis-ard, 2023-03-15:
This looks normal, you can try to run the preview_depth example to see if it is normal.