ArduCAM/Arducam_tof_camera #37: ToFcamera can't work on Xavier NX

Allbeargntoooo, 2023-03-12:
Hi !

I follow the instructions install SDK on my Jetson Xavier NX(Jetpack 5.0.2), but result like this photo.

[image]

I also install SDK on my Jetson Nano, ToFcamera can work.

dennis-ard, 2023-03-13:
Hi~,
Judging from this picture, the TOF camera seems to have no external power supply. You can check whether the external power supply of the TOF camera is connected correctly, and then restart the camera program to see if the problem is improved.
Best Regard.

ahtonen, 2023-05-31:
I had exactly this issue with Jetson Nano. Turned out that I used the wrong side of the 40-pin connector... had signal pins instead of +5V and GND (sic).

dennis-ard, 2024-03-27:
> Hi, 你好 Same problem is happening to me, **My power pins are connected correctly**. I am using a Jetson Orin Nano Developer kit.同样的问题发生在我身上，我的电源引脚连接正确。我正在使用 Jetson Orin Nano 开发套件。 What are the reason except power which can cause this?除了权力之外，还有什么原因会导致这种情况？ Please help me. 请帮帮我。 ![WhatsApp Image 2024-03-27 at 11 33 11_c906d8a3](https://private-user-images.githubusercontent.com/54736857/317149039-4beadee5-bccd-4538-a55a-973aae62fc25.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE1MjU5ODAsIm5iZiI6MTcxMTUyNTY4MCwicGF0aCI6Ii81NDczNjg1Ny8zMTcxNDkwMzktNGJlYWRlZTUtYmNjZC00NTM4LWE1NWEtOTczYWFlNjJmYzI1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI3VDA3NDgwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg0ZWVjMTEzNDYzNmNmNjdmZjFkNmUyMzNmNmM1Mzg0NmQzYTE3YTA1MTJkMDY5NmJkNWI1NTg4NTcyOTA5NjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ECr0aReNW4mtPuEYPPgP3xX6wSqzT0l58imKL75a-nU) ![WhatsApp Image 2024-03-27 at 11 33 15_b088d64b](https://private-user-images.githubusercontent.com/54736857/317149052-1b862e3f-1ab2-496d-b6d7-9c034bcda96f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE1MjU5ODAsIm5iZiI6MTcxMTUyNTY4MCwicGF0aCI6Ii81NDczNjg1Ny8zMTcxNDkwNTItMWI4NjJlM2YtMWFiMi00OTZkLWI2ZDctOWMwMzRiY2RhOTZmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI3VDA3NDgwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZjOWI4ODc2MDU2ZjY4N2ZmOGRlNjlmYzg1NDhkZjcwMjU2MDg0ZmMzNGIwMTgyNzk1M2NmZDMzMTlmZjU5NzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.BXYJbJ2U95JL41nEAMwUYq5UJ_tjP7yqyCJVxFq9an0)
hello,
This seems even stranger. Could you please tell me whether the Jetson Orin Nano Developer kit is connected to other cameras? Also, I need you to provide me with the complete dmesg kernel log.
```shell
sudo dmesg
```

Programmer-Bose, 2024-03-27:
No Jetson Orin Nano is not connected with any other camera. **is it necessary to build opencv with cuda** for this to work?
 **dmesg kernel log:**
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```
sudo dmesg
[sudo] password for obseract: 
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd421]
[    0.000000] Linux version 5.10.120-tegra (arducam_006@arducam-server-006) (aarch64-linux-gcc.br_real (Buildroot 2020.08) 9.3.0, GNU ld (GNU Binutils) 2.33.1) #1 SMP PREEMPT Wed Aug 9 15:29:32 CST 2023
[    0.000000] OF: fdt: memory scan node memory@80000000, reg size 16,
[    0.000000] OF: fdt:  - 80000000 ,  c0000000
[    0.000000] Machine model: NVIDIA Orin Nano Developer Kit
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: RTPROP=0x266eb1f18 SMBIOS=0xffff0000 SMBIOS 3.0=0x247620000 MEMATTR=0x24844f018 ESRT=0x24751a918 RNG=0x24883f318 MEMRESERVE=0x248450c18 
[    0.000000] efi: seeding entropy pool
[    0.000000] esrt: Reserving ESRT space from 0x000000024751a918 to 0x000000024751a950.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000025a800000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x0000000273ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x00000000fffdffff]
[    0.000000]   node   0: [mem 0x00000000fffe0000-0x00000000ffffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x0000000246f16fff]
[    0.000000]   node   0: [mem 0x0000000246f17000-0x00000002471aefff]
[    0.000000]   node   0: [mem 0x00000002471af000-0x000000024756ffff]
[    0.000000]   node   0: [mem 0x0000000247570000-0x000000024762ffff]
[    0.000000]   node   0: [mem 0x0000000247630000-0x00000002482fffff]
[    0.000000]   node   0: [mem 0x0000000248300000-0x000000024839ffff]
[    0.000000]   node   0: [mem 0x00000002483a0000-0x00000002483bffff]
[    0.000000]   node   0: [mem 0x00000002483c0000-0x00000002483dffff]
[    0.000000]   node   0: [mem 0x00000002483e0000-0x00000002483effff]
[    0.000000]   node   0: [mem 0x00000002483f0000-0x000000024841ffff]
[    0.000000]   node   0: [mem 0x0000000248420000-0x000000024845ffff]
[    0.000000]   node   0: [mem 0x0000000248460000-0x000000024847ffff]
[    0.000000]   node   0: [mem 0x0000000248480000-0x000000024858ffff]
[    0.000000]   node   0: [mem 0x0000000248590000-0x000000024876ffff]
[    0.000000]   node   0: [mem 0x0000000248770000-0x000000024882ffff]
[    0.000000]   node   0: [mem 0x0000000248830000-0x00000002488dffff]
[    0.000000]   node   0: [mem 0x00000002488e0000-0x000000024892ffff]
[    0.000000]   node   0: [mem 0x0000000248930000-0x0000000248b5ffff]
[    0.000000]   node   0: [mem 0x0000000248b60000-0x0000000248beffff]
[    0.000000]   node   0: [mem 0x0000000248bf0000-0x0000000248c3ffff]
[    0.000000]   node   0: [mem 0x0000000248c40000-0x0000000248c6ffff]
[    0.000000]   node   0: [mem 0x0000000248c70000-0x0000000248d0ffff]
[    0.000000]   node   0: [mem 0x0000000248d10000-0x0000000248d5ffff]
[    0.000000]   node   0: [mem 0x0000000248d60000-0x0000000248dfffff]
[    0.000000]   node   0: [mem 0x0000000248e00000-0x00000002492dffff]
[    0.000000]   node   0: [mem 0x00000002492e0000-0x00000002494bffff]
[    0.000000]   node   0: [mem 0x00000002494c0000-0x000000026ab2ffff]
[    0.000000]   node   0: [mem 0x000000026ab30000-0x000000026ab7ffff]
[    0.000000]   node   0: [mem 0x000000026ab80000-0x000000026ea0ffff]
[    0.000000]   node   0: [mem 0x000000026ea10000-0x000000026ea1ffff]
[    0.000000]   node   0: [mem 0x0000000272000000-0x0000000273ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x0000000273ffffff]
[    0.000000] On node 0 totalpages: 2034208
[    0.000000]   DMA zone: 8192 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 524288 pages, LIFO batch:63
[    0.000000]   Normal zone: 23808 pages used for memmap
[    0.000000]   Normal zone: 1509920 pages, LIFO batch:63
[    0.000000] On node 0, zone Normal: 480 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 32 pages/cpu s90328 r8192 d32552 u131072
[... 1146 lines cut ...]
[  196.341786] tegra-camrtc-capture-vi tegra-capture-vi: err_rec: attempting to reset the capture channel
[  196.352504] (NULL device *): vi_capture_control_message: NULL VI channel received
[  196.360246] t194-nvcsi 13e40000.host1x:nvcsi@15a00000: csi5_stream_close: Error in closing stream_id=0, csi_port=0
[  196.370925] (NULL device *): vi_capture_control_message: NULL VI channel received
[  196.378660] t194-nvcsi 13e40000.host1x:nvcsi@15a00000: csi5_stream_open: VI channel not found for stream- 0 vc- 0
[  196.389541] tegra-camrtc-capture-vi tegra-capture-vi: err_rec: successfully reset the capture channel
[... cut ...]
[  205.436883] mmc1: Applying tuning correction
[  205.440516] mmc1: Tap value after applying correction 77
[  213.180746] mmc1: Applying tuning correction
[  213.180770] mmc1: Tap value after applying correction 77
[  213.408409] mmc1: Applying tuning correction
[  213.408427] mmc1: Tap value after applying correction 77
```
-----------------------------------------------------------------------------------------------------------------------------------------------------
[image]
[image]
Please help!!