ArduCAM/Arducam_tof_camera #39: Unable to install Arducam depth sensor dependencies on Raspberry Pi 4B

Jayyo, 2023-03-24:
I am currently trying to install the dependencies for using the Arducam depth sensor module with Python on my Raspberry Pi 4B, but I am encountering some issues.

I was able to view the depth image and greyscaled monitor using the compile and run command, but when I try to install the dependencies using the ./Install_dependencies_python.sh command, my terminal gets stuck. I have also tried to manually install the dependencies as per the instructions on the Arducam GitHub page, but I get stuck at ./install_pivariety_pkgs.sh -p kernal_driver.

I am using a Raspberry Pi 4B with kernel version 6.1.19-v8+ and the Bullseye-arm64 operating system. I opened the kernel driver link.txt file, but I cannot access the URL mentioned in the file.

I would greatly appreciate any guidance or support on how to install the necessary dependencies for using the Arducam depth sensor module with Python on my Raspberry Pi 4B.

Thank you for your time and assistance.

Environment:

Raspberry Pi 4B
Kernel version 6.1.19-v8+
Bullseye-arm64 operating system
Steps to Reproduce:

Download and install Arducam depth sensor module for Raspberry Pi
Try to install dependencies using ./Install_dependencies_python.sh command
Terminal gets stuck
Try to manually install dependencies, get stuck at ./install_pivariety_pkgs.sh -p kernal_driver
Expected Result:
Successfully install dependencies and use Arducam depth sensor module with Python on Raspberry Pi 4B.

Actual Result:
Unable to install dependencies, getting stuck at ./Install_dependencies_python.sh and ./install_pivariety_pkgs.sh -p kernal_driver.

[image]
[image]

Jayyo, 2023-03-25:
I was able to I was able to solve the issue by installing the required packages using the command 'pip3 install opencv-python ArducamDepthCamera'. However, when I tried to view the camera feed using 'VideoCapture(0)' in OpenCV, it only displayed amplitude data instead of the original camera frame. Is there a way to view the camera feed in its original format, like a normal camera

BeamNC, 2023-03-30:
I'm facing the same issue. Have you managed to fix this?

flatypus, 2025-07-03:
[image]

It works for me! Running Ubuntu 24.04 LTS on a Raspberry Pi Compute Module 4; I only needed to run Install_Dependencies.sh, which took ~10 minutes but worked after reboot. The example python code worked fine. 

@Jayyo what do you mean by viewing it as a normal camera? I'm not sure if this ToF supports both streams...people online seem to have better luck just having two cameras. If you want to check out the classes themselves, the [https://pypi.org/project/ArducamDepthCamera/0.1.21/#ArducamDepthCamera-0.1.21-cp313-cp313t-manylinux_2_27_x86_64.whl](.pyi stubs) exist here