ArduCAM/Arducam_tof_camera #41: pcl_preview build error

jenyak, 2023-04-03:
Both examples
```
  cd Arducam_tof_camera/example
  mkdir build && cd build
  cmake .. && make
```
  
  and 
```
  cd Arducam_tof_camera/pcl_preview
  mkdir build && cd build
  cmake ..
  make
```
fail with same error 
```
[ 78%] Building CXX object pcl_preview/CMakeFiles/preview_pointcloud.dir/preview_pointcloud.cpp.o
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp: In function ‘int main()’:
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:85:48: error: ‘simpleVis’ was not declared in this scope
   85 |     pcl::visualization::PCLVisualizer viewer = simpleVis(cloud_ptr);
      |                                                ^~~~~~~~~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:136:19: error: base operand of ‘->’ has non-pointer type ‘pcl::visualization::PCLVisualizer’
  136 |             viewer->updatePointCloud<pcl::PointXYZ>(cloud_ptr, "sample cloud");
      |                   ^~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:136:51: error: expected primary-expression before ‘>’ token
  136 |             viewer->updatePointCloud<pcl::PointXYZ>(cloud_ptr, "sample cloud");
      |                                                   ^
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:136:53: warning: left operand of comma operator has no effect [-Wunused-value]
  136 |             viewer->updatePointCloud<pcl::PointXYZ>(cloud_ptr, "sample cloud");
      |                                                     ^~~~~~~~~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:137:19: error: base operand of ‘->’ has non-pointer type ‘pcl::visualization::PCLVisualizer’
  137 |             viewer->spinOnce(100);
      |                   ^~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:149:18: error: jump to case label
  149 |             case 'q':
      |                  ^~~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:144:21: note:   crosses initialization of ‘tm* nowtime’
  144 |                 tm *nowtime = localtime(&t);
      |                     ^~~~~~~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:143:29: note:   crosses initialization of ‘time_t t’
  143 |                 std::time_t t = std::time(0);
      |                             ^
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:151:23: error: base operand of ‘->’ has non-pointer type ‘pcl::visualization::PCLVisualizer’
  151 |                 viewer->close();
      |                       ^~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:158:18: error: jump to case label
  158 |             case 'd':
      |                  ^~~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:144:21: note:   crosses initialization of ‘tm* nowtime’
  144 |                 tm *nowtime = localtime(&t);
      |                     ^~~~~~~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:143:29: note:   crosses initialization of ‘time_t t’
  143 |                 std::time_t t = std::time(0);
      |                             ^
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:159:29: error: redeclaration of ‘time_t t’
  159 |                 std::time_t t = std::time(0);
      |                             ^
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:143:29: note: ‘time_t t’ previously declared here
  143 |                 std::time_t t = std::time(0);
      |                             ^
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:160:21: error: redeclaration of ‘tm* nowtime’
  160 |                 tm *nowtime = localtime(&t);
      |                     ^~~~~~~
/home/pi/Arducam_tof_camera/pcl_preview/preview_pointcloud.cpp:144:21: note: ‘tm* nowtime’ previously declared here
  144 |                 tm *nowtime = localtime(&t);
      |                     ^~~~~~~
make[2]: *** [pcl_preview/CMakeFiles/preview_pointcloud.dir/build.make:82: pcl_preview/CMakeFiles/preview_pointcloud.dir/preview_pointcloud.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:335: pcl_preview/CMakeFiles/preview_pointcloud.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

```

On fresh RPI 4b 4GB
```
$ uname -a
Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
```

shamoilkazmi, 2023-04-04:
I am also getting the same issue

p-ryals, 2023-04-07:
This is happening for me as well, on a brand new rpi install. I found references to the [function](https://github.com/PointCloudLibrary/pcl/blob/master/doc/tutorials/content/sources/pcl_visualizer/pcl_visualizer_demo.cpp), but no matter what I've tried, it just won't compile. I'm not extremely good with C, though, so I'm not sure what I'm missing.

shamoilkazmi, 2023-04-07:
I have compiled pcl-pcl-1.12.0 but facing same issue.

shamoilkazmi, 2023-04-07:
Actually there are error in the code preview_pointcloud.cpp

shamoilkazmi, 2023-04-07:
I have tweaked the code and got success in compilatin :)
#include <iostream>
#include <pcl/io/pcd_io.h>
#include <pcl/point_types.h>
#include <pcl/visualization/cloud_viewer.h>
#include <opencv2/core.hpp>
#include <opencv2/imgproc.hpp>
#include <opencv2/highgui.hpp>
#include <ctime>
#include "ArducamTOFCamera.hpp"

#define MAX_DISTANCE 4
using namespace Arducam;

boost::shared_ptr<pcl::visualization::PCLVisualizer> simpleVis(pcl::PointCloud<pcl::PointXYZ>::ConstPtr cloud)
{
    boost::shared_ptr<pcl::visualization::PCLVisualizer> viewer(new pcl::visualization::PCLVisualizer("3D Viewer"));
    viewer->addCoordinateSystem(0.01);
    viewer->setCameraPosition(0, 0, 0, 0, 0, 0, 0, 0, -1);
    viewer->setBackgroundColor(0, 0, 0);
    viewer->addPointCloud<pcl::PointXYZ>(cloud, "sample cloud");
    viewer->initCameraParameters();
    return (viewer);
}

void getPreview(uint8_t *preview_ptr, float *depth_image_ptr, float *amplitude_image_ptr)
{
    auto len = 240 * 180;
    for (int i = 0; i < len; i++)
    {
        uint8_t mask = *(amplitude_image_ptr + i) > 30 ? 254 : 0;
        float depth = ((1 - (*(depth_image_ptr + i) / MAX_DISTANCE)) * 255);
        uint8_t pixel = depth > 255 ? 255 : depth;
        *(preview_ptr + i) = pixel & mask;
    }
}

int main()
{
    ArducamTOFCamera tof;
    ArducamFrameBuffer *frame;
    char buff[60];
    float *depth_ptr;
    float *amplitude_ptr;
    uint8_t *preview_ptr = new uint8_t[43200];
    if (tof.init(Connection::CSI))
    {
        std::cerr << "initialization failed" << std::endl;
        exit(-1);
    }
    if (tof.start())
    {
        std::cerr << "Failed to start camera" << std::endl;
        exit(-1);
    }
    tof.setControl(ControlID::RANGE, MAX_DISTANCE);

    pcl::PointCloud<pcl::PointXYZ>::Ptr cloud_ptr(new pcl::PointCloud<pcl::PointXYZ>);

    cloud_ptr->points.push_back(pcl::PointXYZ(10, 10, 4));
    cloud_ptr->width = cloud_ptr->size();
    cloud_ptr->height = 1;
    cloud_ptr->is_dense = true;
    vtkObject::GlobalWarningDisplayOff();

    boost::shared_ptr<pcl::visualization::PCLVisualizer> viewer = simpleVis(cloud_ptr);

    const float fx = 240 / (2 * tan(0.5 * M_PI * 64.3 / 180));
    const float fy = 180 / (2 * tan(0.5 * M_PI * 50.4 / 180));
    for (;;)
    {
        frame = tof.requestFrame(200);

        if (frame != nullptr)
        {
            depth_ptr = (float *)frame->getData(FrameType::DEPTH_FRAME);
            amplitude_ptr = (float *)frame->getData(FrameType::AMPLITUDE_FRAME);
            getPreview(preview_ptr, depth_ptr, amplitude_ptr);
            cv::Mat result_frame(180, 240, CV_8U, preview_ptr);
            cv::Mat amplitude_frame(180, 240, CV_32F, amplitude_ptr);

           
            cv::Mat depth_frame(180, 240, CV_32F, depth_ptr);

            pcl::PointCloud<pcl::PointXYZ>::Ptr current_cloud(new pcl::PointCloud<pcl::PointXYZ>);
            current_cloud->height = 180;
            current_cloud->width = 240;
            current_cloud->points.resize(current_cloud->height * current_cloud->width);

            for (int v = 0; v < depth_frame.rows; ++v)
            {
                for (int u = 0; u < depth_frame.cols; ++u)
                {
                    float z = depth_frame.at<float>(v, u);
                    float x = (u - depth_frame.cols / 2.0) * z / fx;
                    float y = (v - depth_frame.rows / 2.0) * z / fy;
                    current_cloud->points[v * depth_frame.cols + u].x = x;
                    current_cloud->points[v * depth_frame.cols + u].y = y;
                    current_cloud->points[v * depth_frame.cols + u].z = z;
                }
            }

            viewer->updatePointCloud<pcl::PointXYZ>(current_cloud, "sample cloud");

            viewer->spinOnce(1);
            pcl::io::savePCDFileASCII("test_pcd.pcd", *current_cloud);

            cv::imshow("Preview", result_frame);
            cv::imshow("Amplitude", amplitude_frame);
            cv::waitKey(1);

            tof.releaseFrame(frame);
        }
    }

    tof.stop();
    delete[] preview_ptr;

    return 0;
}

p-ryals, 2023-04-07:
Fantastic! Your copy/paste didn't come over cleanly, but I diff'd between the old and new and figured out what was missing. I was able to compile and run it as well. Thanks so much!

shamoilkazmi, 2023-04-07:
[preview_pointcloud.zip](https://github.com/ArduCAM/Arducam_tof_camera/files/11181855/preview_pointcloud.zip)
Here you go I have attached the file.

maureen-sun-0812, 2024-08-14:
Hello, thank you for doing this! I'm getting similar errors, and tried using your modify script, but it won't build anymore. Currently, the original .cpp file builds, but I'm getting this error, and I'm wondering if you ever got the same. 

`raspi@raspi:~/Arducam_tof_camera/build/pcl_preview$ ./preview_pointcloud
open camera with (240x180) with range 4000
[ WARN:0] global ./modules/highgui/src/window.cpp (703) createTrackbar UI/Trackbar(confidence@preview): Using 'value' pointer is unsafe and deprecated. Use NULL as value pointer. To fetch trackbar value setup callback.
frame: (240x180)
Segmentation fault (core dumped)`