ArduCAM/Arducam_tof_camera #47: Raspberry Pi Compile Error

jhsong999, 2023-05-15:
Hello, I trying Connect ToF Camera with Raspberry pi.
but error occurs during the compile && run stage.

I Connect ToF Camera and Download Clone "https://github.com/ArduCAM/Arducam_tof_camera"
Next time, Install dependencies after reboot.
Then next time, I tried to compile, but an error occurred.
The error contents are as follows.
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ
[30%] Building CXX object cpp/CMakeFiles/preview_depth.dir/preview_depth.cpp.o
/home/pi/Arducam_tof_camera/example/cpp/preview_depth.cpp:2:28: fatal error: opencv2/core.hpp: No such file or directory
                  
compilation terminated.
cpp/CMakeFiles/preview_depth.dir/build.make:62: recipe for target 'cpp/CMakeFiles/preview_depth.dir/preview_depth.cpp.o' failed
make[2]: *** [cpp/CMakeFiles/preview_depth.dir/preview_depth.cpp.o] Error 1
CMakeFiles/Makefile2:146: recipe for target 'cpp/CMakeFiles/preview_depth.dir/all' failed
make[1]: *** [cpp/CMakeFiles/preview_depth.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Compilation error, please check whether /dev/video0 exists, or check whether the driver and sdk are installed successfully!
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ

It was confirmed that the path of the opencv2 code was different, so I tried to fix it directly with the link, but I kept getting new errors.

It is not possible to determine exactly whether it is a problem with the opencv2 path or the /dev/video0 camera connection that follows.

I'd like to ask if you can suggest some solutions. Thank you!

jhsong999, 2023-05-16:
Hello, Thank you for responding to my response.
I'd like to ask you one more question before proceeding with the work.
[(https://docs.arducam.com/Raspberry-Pi-Camera/Tof-camera/Getting-Started/)]
I was referring to this link and working on it.
Raspberry Pi used in this link is version 4.
Can I test the same with Raspberry Pi 3 by any chance?
Can Raspberry Pi 3 test the same camera?
The Raspberry Pi version I'm using is "Raspberry Pi 3 Model B V1.2"
Thank you!

dennis-ard, 2023-05-16:
You're welcome!
Yes, it is possible to test the TOF camera module with your Raspberry Pi 3 Model B V1.2.

jhsong999, 2023-05-16:
Hello, I reinstalled opencv and compiled it again.
However, a new error occurred.
Upload the build and error log.
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ
pi@raspberrypi:~/Downloads/Arducam_tof_camera $ ./compile.sh
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Downloads/Arducam_tof_camera/example/build
Scanning dependencies of target test_c
[ 10%] Building C object c/CMakeFiles/test_c.dir/test.c.o
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c: In function ‘main’:
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:21:9: warning: implicit declaration of function ‘init’ [-Wimplicit-function-declaration]
     if (init(tof,CSI,0))
         ^~~~
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:23:9: warning: implicit declaration of function ‘start’ [-Wimplicit-function-declaration]
     if (start(tof,DEPTH_FRAME))
         ^~~~~
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:30:18: warning: implicit declaration of function ‘requestFrame’ [-Wimplicit-function-declaration]
     if ((frame = requestFrame(tof,200)) != 0x00){
                  ^~~~~~~~~~~~
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:30:16: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     if ((frame = requestFrame(tof,200)) != 0x00){
                ^
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:31:18: warning: implicit declaration of function ‘getFormat’ [-Wimplicit-function-declaration]
         format = getFormat(frame,DEPTH_FRAME);
                  ^~~~~~~~~
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:31:16: error: incompatible types when assigning to type ‘FrameFormat {aka struct <anonymous>}’ from type ‘int’
         format = getFormat(frame,DEPTH_FRAME);
                ^
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:35:20: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
         if ((frame = requestFrame(tof,200)) != 0x00)
                    ^
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:37:33: warning: implicit declaration of function ‘getDepthData’ [-Wimplicit-function-declaration]
             depth_ptr = (float*)getDepthData(frame);
                                 ^~~~~~~~~~~~
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:39:37: warning: implicit declaration of function ‘getAmplitudeData’ [-Wimplicit-function-declaration]
             amplitude_ptr = (float*)getAmplitudeData(frame);
                                     ^~~~~~~~~~~~~~~~
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:41:13: warning: implicit declaration of function ‘releaseFrame’ [-Wimplicit-function-declaration]
             releaseFrame(tof,frame);  
             ^~~~~~~~~~~~
/home/pi/Downloads/Arducam_tof_camera/example/c/test.c:45:9: warning: implicit declaration of function ‘stop’ [-Wimplicit-function-declaration]
     if (stop(tof))
         ^~~~
c/CMakeFiles/test_c.dir/build.make:62: recipe for target 'c/CMakeFiles/test_c.dir/test.c.o' failed
make[2]: *** [c/CMakeFiles/test_c.dir/test.c.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'c/CMakeFiles/test_c.dir/all' failed
make[1]: *** [c/CMakeFiles/test_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Compilation error, please check whether /dev/video0 exists, or check whether the driver and sdk are installed successfully!
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ

It is presumed to be a compilation error related to CMake, but the exact cause has not been found so far.
Can you look at the following error log and give me some advice? Thank you!

dennis-ard, 2023-05-16:
yes, I am happy to inform you that you are about to succeed, just pull the latest code from the repos and compile it.

jhsong999, 2023-05-16:
I'm sorry, but I didn't quite understand the instructions to get the latest code and compile it.

The code currently being compiled by Raspberry Pi is from the last version of the repository.

Are you saying that the compilation will fail simply because the code is old?

Could you please post a link where I can download the last version of the code?

Excuse me while you're busy.

dennis-ard, 2023-05-16:
I just updated the code of the repos,you can check whether the commit is 6d5be4219b0475798217286e73e17d917678a392.

jhsong999, 2023-05-16:
Thank you! I got the code from the latest store and compiled it.
However, a new error occurred again.
New error code.
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ
pi@raspberrypi:~/Downloads/Arducam_tof_camera $ ./compile.sh
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Downloads/Arducam_tof_camera/example/build
[ 20%] Built target test_c
Scanning dependencies of target preview_depth
[ 30%] Building CXX object cpp/CMakeFiles/preview_depth.dir/preview_depth.cpp.o
/home/pi/Downloads/Arducam_tof_camera/example/cpp/preview_depth.cpp: In function ‘int main()’:
/home/pi/Downloads/Arducam_tof_camera/example/cpp/preview_depth.cpp:121:13: error: ‘applyColorMap’ is not a member of ‘cv’
             cv::applyColorMap(result_frame, result_frame, cv::COLORMAP_JET);
             ^~
/home/pi/Downloads/Arducam_tof_camera/example/cpp/preview_depth.cpp:121:59: error: ‘COLORMAP_JET’ is not a member of ‘cv’
             cv::applyColorMap(result_frame, result_frame, cv::COLORMAP_JET);
                                                           ^~
cpp/CMakeFiles/preview_depth.dir/build.make:62: recipe for target 'cpp/CMakeFiles/preview_depth.dir/preview_depth.cpp.o' failed
make[2]: *** [cpp/CMakeFiles/preview_depth.dir/preview_depth.cpp.o] Error 1
CMakeFiles/Makefile2:146: recipe for target 'cpp/CMakeFiles/preview_depth.dir/all' failed
make[1]: *** [cpp/CMakeFiles/preview_depth.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Compilation error, please check whether /dev/video0 exists, or check whether the driver and sdk are installed successfully!
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ

An error appears to be outputting for the cv structure.

Can you give me some advice on the problem of the following error? Thank you always!

jhsong999, 2023-05-16:
For additional information, the version of opencv you installed is 2.4.9.1.

Could it have anything to do with this?

dennis-ard, 2023-05-16:
I checked the [documentation of OpenCV 2.4.9](https://docs.opencv.org/2.4.9/modules/contrib/doc/facerec/colormaps.html?highlight=colormap) and found that it is there,so if it is normal, there should not be such a compilation error. I am sorry that this code I only tested on OpenCV 4.0+ version. Maybe you can try to upgrade the version of OpenCV, or just comment this statement. This The purpose of the segment statement is to map the grayscale values of the image to JET.

dennis-ard, 2023-05-16:
Can you use `unam -a` to provide your system version, Install_dependencies.sh, this script only installs opencv through the following command, and generally installs OpenCV 4.5+ version in the system version of  **bullseye**  .
```shell
sudo apt update
sudo apt install libopencv-dev
```
You can manually search to see if there is a higher version of opencv and download opencv.
```Shell
apt search libopencv-dev
```
The opencv library is in namespace cv.

If it is not possible to manually download a higher version of opencv, it is recommended that you flash the latest bullseye system to run the Arducam TOF camera.
best wishes

jhsong999, 2023-05-16:
This is the result of using the "uname -a" command.

Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

And I downloaded OpenCv again using the command below.

sudo apt update
sudo apt install libopencv-dev

But the same version was 2.4.9.1.

I also manually searched for higher versions, and the maximum version detected was 2.4.9.1.
In this case, should I try to install it manually?
Let's try to install version 4.0 or later of OpenCv.

dennis-ard, 2023-05-17:
It looks like your Raspberry Pi version is buster, and the system version required for the camera to run is bullseye.
Run `lsb_release -a`, you can view your codenam. It is recommended that you update your system to bullseye.

jhsong999, 2023-05-17:
Results for lsb_release -a.

ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ
No LSB modlues are available.
Distributor ID: Raspbian
Description:     Raspbian GNU/Linux 9.13 (stretch)
Release:            9.13
Codename:       stretch
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ

I understood that Raspberry Pi's version is an "buster", so I need to update it to "bullseye", the camera operating environment.

Is my understanding correct?

If it's right, I'm going to upload it as bullseye on my new SD card.

dennis-ard, 2023-05-17:
yep,that's right.

jhsong999, 2023-05-17:
I updated Raspberry Pi os new and it was compiled normally!

Thank you so much for responding to so many requests!

dennis-ard, 2023-05-17:
It is my pleasure to serve you!