ArduCAM/Arducam_tof_camera #49: ToF Sensor Python Build

jhsong999, 2023-05-17:
Hello, Following the compilation issue, I asked for another problem.

I'm trying to proceed with the camera build using Python, but the following problem has occurred.

./Install_dependencies_python.sh

I'm trying to run the installation by typing the following command, but it won't last for a long time.

I also take a picture of the terminal window and send it to you.

[image]

dennis-ard, 2023-05-17:
I'm sorry that I made an error after a certain script update before, and it has been fixed in this pr #50  now

wenj5, 2023-05-20:
i m sorry, i v tried the new one, but still can not work...