ArduCAM/Arducam_tof_camera #60: Failed to install TOF camera on Jetson Nano

aoblue2547, 2023-10-12:
I want to install Arducam ToF Camera on Jetson Nano (L4T 32.6.1)
but failed to complie.

What should I do?

First, I ran Install_dependencies.sh
```
Get:1 file:/var/cuda-repo-l4t-10-2-local  InRelease
Ign:1 file:/var/cuda-repo-l4t-10-2-local  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease
Ign:3 file:/var/visionworks-sfm-repo  InRelease
Get:4 file:/var/visionworks-tracking-repo  InRelease
Ign:4 file:/var/visionworks-tracking-repo  InRelease 
Get:5 file:/var/cuda-repo-l4t-10-2-local  Release [564 B]                   
Get:6 file:/var/visionworks-repo  Release [2,001 B]                            
Get:7 file:/var/visionworks-sfm-repo  Release [2,005 B]                        
Get:8 file:/var/visionworks-tracking-repo  Release [2,010 B]                   
Get:5 file:/var/cuda-repo-l4t-10-2-local  Release [564 B]                      
Get:6 file:/var/visionworks-repo  Release [2,001 B]                            
Get:7 file:/var/visionworks-sfm-repo  Release [2,005 B]                        
Get:8 file:/var/visionworks-tracking-repo  Release [2,010 B]                   
Hit:9 https://repo.download.nvidia.com/jetson/common r32.6 InRelease           
Hit:10 https://repo.download.nvidia.com/jetson/t210 r32.6 InRelease            
Hit:11 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Get:12 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [83.3 kB]
Get:15 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 DEP-11 Metadata [290 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 DEP-11 Metadata [298 kB]
Get:21 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main arm64 DEP-11 Metadata [5,372 B]
Get:22 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe arm64 DEP-11 Metadata [10.0 kB]
Get:23 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 DEP-11 Metadata [70.9 kB]
Get:24 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 DEP-11 Metadata [57.4 kB]
Fetched 993 kB in 13s (78.0 kB/s)                                              
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package arducam-config-parser-dev
E: Unable to locate package arducam-usb-sdk-dev
E: Unable to locate package arducam-tof-sdk-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.10.2-1ubuntu2.18.04.2).
curl is already the newest version (7.58.0-2ubuntu3.24).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--2023-10-12 16:18:37--  https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/install_full.sh
Resolving github.com (github.com)... 20.27.177.113
Connecting to github.com (github.com)|20.27.177.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/ebdc0c54-a520-42e6-b4ef-e241b3acfd59?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231012%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231012T071915Z&X-Amz-Expires=300&X-Amz-Signature=e04626c5745c6565d2768dbfffdf5af7b58be5e90962bebf2a795d8aa5b8e060&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dinstall_full.sh&response-content-type=application%2Foctet-stream [following]
--2023-10-12 16:18:38--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/ebdc0c54-a520-42e6-b4ef-e241b3acfd59?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231012%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231012T071915Z&X-Amz-Expires=300&X-Amz-Signature=e04626c5745c6565d2768dbfffdf5af7b58be5e90962bebf2a795d8aa5b8e060&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dinstall_full.sh&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.111.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4560 (4.5K) [application/octet-stream]
Saving to: ‘install_full.sh’

install_full.sh     100%[===================>]   4.45K  --.-KB/s    in 0s      

2023-10-12 16:18:38 (15.1 MB/s) - ‘install_full.sh’ saved [4560/4560]

./install_full.sh: line 19: warning: command substitution: ignored null byte in input
NVIDIA Jetson Nano Developer Kit
[... 76 lines cut ...]
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable exact version "10.2") 
-- Found OpenCV: /usr (found version "4.5.5") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/robocon/Arducam_tof_camera/jetson/build
Scanning dependencies of target jetson_preview
[ 50%] Building CXX object CMakeFiles/jetson_preview.dir/jetson_preview.cpp.o
/home/robocon/Arducam_tof_camera/jetson/jetson_preview.cpp:1:10: fatal error: ArducamTOFCamera.hpp: No such file or directory
 #include "ArducamTOFCamera.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/jetson_preview.dir/build.make:82: recipe for target 'CMakeFiles/jetson_preview.dir/jetson_preview.cpp.o' failed
make[2]: *** [CMakeFiles/jetson_preview.dir/jetson_preview.cpp.o] Error 1
CMakeFiles/Makefile2:95: recipe for target 'CMakeFiles/jetson_preview.dir/all' failed
make[1]: *** [CMakeFiles/jetson_preview.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2
Compilation error, please check whether the driver and sdk are installed successfully!
```

dennis-ard, 2023-10-12:
Have you downloaded the arducam installation package before? If you have installed the arducam installation package before, the arducam_pp warehouse may not be added. You can manually add the warehouse source and download arducam-config-parser-dev arducam-usb-sdk-dev arducam-tof-sdk-dev,the operation is as follows:
```Shell
curl -s --compressed "https://arducam.github.io/arducam_ppa/KEY.gpg" | sudo apt-key add -
sudo curl -s --compressed -o /etc/apt/sources.list.d/arducam_list_files.list "https://arducam.github.io/arducam_ppa/arducam_list_files.list"
sudo apt update
sudo apt install -y arducam-config-parser-dev arducam-usb-sdk-dev arducam-tof-sdk-dev
```
Then please recompile and run

aoblue2547, 2023-10-12:
compile finished successfully.
Thank you!

dennis-ard, 2023-10-12:
Cool, that's good to hear.
You're welcome!