ArduCAM/Arducam_tof_camera #61: Examples not running getting: "Could not open device node /dev/video0"

HIghlandRebel, 2023-10-12:
RPi 4B with fresh install of Bullseye. Followed the instructions at https://docs.arducam.com/Raspberry-Pi-Camera/Tof-camera/Getting-Started/ 
When attempting to run examples eg. ./c/test_c or ./compile.sh I'm getting a ***WARNING*** "Could not open device node /dev/video0" message. 
ls /dev/video* returns:
"/dev/video10  /dev/video13  /dev/video16  /dev/video20  /dev/video23
/dev/video11  /dev/video14  /dev/video18  /dev/video21  /dev/video31
/dev/video12  /dev/video15  /dev/video19  /dev/video22"

How do I find & connect to the correct device node?

dennis-ard, 2023-10-13:
This means that your camera driver is not loaded correctly or the camera is not connected properly. Can you provide some information using the following commands?

```Shell
cat /boot/config.txt
```

```Shell
dmesg | grep arducam
```

```Shell
uname -a
```

HIghlandRebel, 2023-10-17:
The cat /boot/config.txt returns:

```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
#camera_auto_detect=0

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

dtoverlay=arducam-pivariety,media-controller=0
```
dmesg | grep arducam did not return anything and uname -a returns:

```
Linux raspberrypi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
```

dennis-ard, 2023-10-18:
hi,
It seems that your Raspberry Pi system version is too new. We only support system versions up to Bullseye.

LaserBorg, 2023-12-31:
Arducam ToF on a Pi5 with same problem here, @dennis-ard when can we expect an update on this issue? best