ArduCAM/Arducam_tof_camera #63: problems with many python example dependencies

ketauk, 2023-11-21:
Hi

After buying your ToF camera I have been trying out your example code, but there are a few things I had to work around to get it to work & thought you might like to know. (I am not a developer but exploring with ChatGPT helping me)

NOTE: I have tried the 32-bit RaspberryPi OS but could not get Python to install, so I am now trying the 64-bit OS.    I did manage to OpenCV to install on the 32-bit OS.

On 32-bit OS it could not finds these dependencies so I was able to get it working if I swapped those for the libraries below:

sudo apt install -y libcblas-dev
sudo apt install -y libqtgui4 
sudo apt install -y libqt4-test

Swapped dependencies:

sudo apt install -y libatlas-base-dev
sudo apt install -y libqt5gui5
sudo apt install -y libqt5test5

On 64-bit system I get the same problem for the dependencies above BUT I also get a new problem with libjasper-dev and this does not seem so easy to solve?

sudo apt install -y libjasper-dev 

Thanks

dennis-ard, 2023-11-21:
Hi,
It seems that you are facing difficulties while deploying the environment. Please don't worry. I need you to provide the system version and the error messages you encountered. Please use the following commands to obtain the required information and provide them to me.

1. `uname -a`
2.  `lsb_release -a`
3.  please provide the model of your Raspberry Pi. 

Recently, there have been significant updates to the Raspberry Pi system, and we are currently working on adapting the drivers to ensure smooth operation on the latest system. If you are not using a Raspberry Pi 5, perhaps you can try an older kernel version, such as the system version from May 2023.
You can click on [this link to download](https://downloads.raspberrypi.com/raspios_armhf/images/raspios_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf.img.xz).
Best regards,

ketauk, 2023-11-21:
Hi

I will provide the info you require below, happy to provide more if useful:

I am using a RaspberryPi 3B+

 $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

 $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

 $ sudo apt install -y libcblas-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libcblas-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libatlas-base-dev:armhf libatlas-base-dev

E: Package 'libcblas-dev' has no installation candidate

Package libqtgui4 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libqtgui4' has no installation candidate

E: Unable to locate package libqt4-test

E: Unable to locate package libjasper-dev

ketauk, 2023-11-21:
Don't know if this is useful for you, but I will mention it as it caused me to spin my wheels for days.

Another thing I had to do, that maybe developers already know, was to modify the memory swap space.

It would run for hours (i.e. 5 or 6hrs) then fail & I would not know why; it was due to an out of memory problem.

Eventually I discovered that I needed to modify the memory swap space and it works, successfully installs and it does it a LOT faster (i.e. 64-bit just did it in minutes not hours)

https://qengineering.eu/install-opencv-on-raspberry-pi.html

# enlarge the boundary (CONF_MAXSWAP)
$ sudo nano /sbin/dphys-swapfile
# give the required memory size (CONF_SWAPSIZE)
$ sudo nano /etc/dphys-swapfile
# reboot afterwards
$ sudo reboot

Then you have to change it back to the normal settings so you don't wear out the storage card.

dennis-ard, 2023-11-21:
Thank you for providing this information. Have you tried running the example while ignoring the errors? In some cases, certain dependencies of an application may not be essential. You can try running the demo and if any errors occur, please provide them to me.

ketauk, 2023-11-21:
Yes.   Today I installed the dependencies I could, which was all except libjasper-dev on the 64-bit OS

I did install these potential alternatives, just in case?     libjpeg-dev libpng-dev

OpenCV installed really quickly 

The examples provided are working on the 64-bit system, so I guess libjasper-dev is not  requirement on the 64-bit OS??

So it is all working (thanks) and I was able to work out the problems with the help of ChatGPT

I am now trying to get Yolo to analyse the data coming out of your ToF camera :)

axel578, 2023-12-03:
I have the exact same issue as listed above

dennis-ard, 2023-12-04:
Don't worry, this can still run the example.