ArduCAM/Arducam_tof_camera #64: Raspberry Pi 5 support

dkjonas, 2023-12-12:
Is the Pi5 supported by the ArducamToF?
Just tried with no success, as the camera wasn't detected.
But I'm not sure if that is due to the Pi5 or it is a software issue. Pi5 seems to use Bookworm, but maybe the camera is only for Bullseye?

dkjonas, 2024-02-08:
Any update on this? Can we expect RPi5 support, or should we find other hardware?

NL-Frank, 2024-02-08:
We are waiting for quite a while now, please see also the forum on this issue and the delays:
https://forum.arducam.com/t/installation-tof-camera-fails-on-bookworm-could-not-open-device-node-dev-video0/5883

dennis-ard, 2024-03-18:
hi,We have nearly finished discussing the TOF drive adapter, but there are still some issues that need to be addressed. Please be patient.