ArduCAM/Arducam_tof_camera #66: Compilation error could not open device node /dev/video0

saipavan4, 2024-01-23:
RPi 4B with 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal
Followed the instructions at https://docs.arducam.com/Raspberry-Pi-Camera/Tof-camera/Getting-Started/
When attempting to run   ./compile.sh 
I'm getting a ***WARNING*** Could not open device node /dev/video0. Please check for permissions.
initialization failed 

after running cat /boot/config.txt
[pi4]
dtoverlay=vc4-fkms-v3d
gpu_mem=128
camera_auto_detect=0
dtoverlay=arducam-pivariety,media-controller=0

How do I solve this issue?

dennis-ard, 2024-01-24:
Hi, Pawan 
The driver for Arducam Pivariety is not integrated into the Ubuntu kernel. Therefore, we currently do not support running the camera on Ubuntu.

saipavan4, 2024-01-26:
Hi, Dennis 
I changed from Ubuntu Kernal to Raspberry Pi Bullseye but I'm still getting an error 
`cat /boot/config.txt`
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=0

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
dtoverlay=arducam-pivariety,media-controller=0

`dmesg | grep arducam`
[    0.080475] platform fe801000.csi: Fixed dependency cycle(s) with /soc/i2c0mux/i2c@1/arducam_pivariety@c
[   10.164226] arducam-pivariety 10-000c: pivariety_read: Reading register 0x103 failed
[   10.164250] arducam-pivariety 10-000c: probe failed

`uname -a`
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

dennis-ard, 2024-02-01:
hi,
This seems to be a case of improperly connected hardware. You can check if the wiring is properly connected or try unplugging and reconnecting the MIPI cable on both the camera and the Raspberry Pi end.

saipavan4, 2024-02-05:
I was able to compile it but when I try to run `  ./c/test_c` or `  ./cpp/preview_depth
  #or
  ./cpp/capture_raw` I get a blank irrespective of the time.
[image]

dennis-ard, 2024-02-06:
Can you use this command to test if the camera has been initialized successfully?
```shell
v4l2-ctl -d0 --set-fmt-video=width=240,height=180,pixelformat='Y12 ' --stream-mmap --stream-count=-1
```

saipavan4, 2024-02-06:
I have the same issue
[image]

dennis-ard, 2024-02-18:
> I have the same issue ![2024-02-06-121903_2560x1440_scrot](https://private-user-images.githubusercontent.com/57193478/302751592-997b6eae-1fea-41f8-8303-b3cb8a86a8ab.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgyMTc5NTcsIm5iZiI6MTcwODIxNzY1NywicGF0aCI6Ii81NzE5MzQ3OC8zMDI3NTE1OTItOTk3YjZlYWUtMWZlYS00MWY4LTgzMDMtYjNjYjhhODZhOGFiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE4VDAwNTQxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU5MDkxODljZWZlOTU2NWNkMjFiNDllM2JhYWU2MzRkMzljMGM5NjU3Y2JiY2VmZmM1MDRhMWVkZGE3ODRiMzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6F-8AElQxweSqPi0HuiyaghQNM0G4twgW4lo3opvuRs)

Could you please provide the kernel logs? It appears that there might be an issue with the driver, but I don't have the precise information. After executing the fetch stream command, could you please share some details using the following command?

```
sudo dmesg | grep
```

saipavan4, 2024-02-19:
Sure here is the information you need 
`sudo dmesg | grep arducam`
[    0.082261] platform fe801000.csi: Fixed dependency cycle(s) with /soc/i2c0mux/i2c@1/arducam_pivariety@c
[    9.203903] arducam-pivariety 10-000c: firmware version: 0x10002
[    9.665759] arducam-pivariety 10-000c: Consider updating driver arducam-pivariety to match on endpoints

dennis-ard, 2024-02-20:
This is not an issue with program dependencies, but rather a problem at the camera driver level. It appears that the driver is loaded correctly, but it is unable to retrieve the image data. Perhaps you can provide me with the complete kernel log by using the command `sudo dmesg` .

saipavan4, 2024-02-20:
Sure this is `sudo dmesg | grep arducam`
```
[    0.080186] platform fe801000.csi: Fixed dependency cycle(s) with /soc/i2c0mux/i2c@1/arducam_pivariety@c
[    9.188397] arducam-pivariety 10-000c: firmware version: 0x10002
[    9.362581] arducam-pivariety 10-000c: Consider updating driver arducam-pivariety to match on endpoints

```
and
`sudo dmesg`

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:23:AC:27 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=48422a20-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 1365140K/2019328K available (11776K kernel code, 2106K rwdata, 3688K rodata, 4160K init, 1077K bss, 129900K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39744 entries in 156 pages
[    0.000000] ftrace: allocated 156 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000332] Console: colour dummy device 80x25
[... 383 lines cut ...]
[ 1132.062058] vc4-drm gpu: [drm] User-defined mode not supported: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x60 0x5
[ 2299.265326] vc4-drm gpu: [drm] User-defined mode not supported: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x60 0x5
[ 2325.429148] usb 1-1.3: new low-speed USB device number 3 using xhci_hcd
[ 2325.537962] usb 1-1.3: New USB device found, idVendor=413c, idProduct=2113, bcdDevice= 1.08
[ 2325.537991] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 2325.538008] usb 1-1.3: Product: Dell KB216 Wired Keyboard
[... cut ...]
[ 2397.787560] vc4-drm gpu: [drm] User-defined mode not supported: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x60 0x5
[ 2399.182154] vc4-drm gpu: [drm] User-defined mode not supported: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x60 0x5
[ 2463.851032] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 2567.061663] vc4-drm gpu: [drm] User-defined mode not supported: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x60 0x5
[ 2612.443883] vc4-drm gpu: [drm] User-defined mode not supported: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x60 0x5

```

dennis-ard, 2024-02-21:
I can only see the log of driver loading here, which appears to be normal. I don't see any relevant information about the command you ran to capture the image. Could you please run the command to capture the image.

`v4l2-ctl -d0 --set-fmt-video=width=240,height=180,pixelformat='Y12 ' --stream-mmap --stream-count=-1`   

then press "Ctrl + C" to end it, and finally use `sudo dmesg` to retrieve the kernel logs?

saipavan4, 2024-02-22:
Kernal logs `sudo dmesg`
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=DC:A6:32:23:AC:27 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=48422a20-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 1365148K/2019328K available (11776K kernel code, 2106K rwdata, 3688K rodata, 4160K init, 1077K bss, 129892K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39744 entries in 156 pages
[    0.000000] ftrace: allocated 156 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000327] Console: colour dummy device 80x25
[    0.000389] printk: console [tty1] enabled
[    0.000447] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000472] pid_max: default: 32768 minimum: 301
[    0.000592] LSM: Security Framework initializing
[    0.000786] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000828] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.002104] cgroup: Disabling memory control group subsystem
[    0.004343] cblist_init_generic: Setting adjustable number of callback queues.
[    0.004362] cblist_init_generic: Setting shift to 2 and lim to 1.
[... 369 lines cut ...]
[   15.671098] NET: Registered PF_BLUETOOTH protocol family
[   15.671104] Bluetooth: HCI device and connection manager initialized
[   15.671121] Bluetooth: HCI socket layer initialized
[   15.671130] Bluetooth: L2CAP socket layer initialized
[   15.671149] Bluetooth: SCO socket layer initialized
[   15.683001] Bluetooth: HCI UART driver ver 2.3
[... cut ...]
[   16.310234] Bluetooth: RFCOMM TTY layer initialized
[   16.310280] Bluetooth: RFCOMM socket layer initialized
[   16.310305] Bluetooth: RFCOMM ver 1.11
[   43.143640] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[   43.148641] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[   43.243481] Console: switching to colour frame buffer device 320x90
[   43.243524] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[  135.646833] hwmon hwmon1: Undervoltage detected!
[  139.678693] hwmon hwmon1: Voltage normalised

```

dennis-ard, 2024-02-23:
This question seems a bit strange. If you need assistance, please contact our support team at support@arducam.com via email. We will provide you with remote support. Thank you.

AnhNhan2803, 2024-03-01:
Have you found a solution to this problem yet?

saipavan4, 2024-03-01:
No, I have not found the solution. I am using an alternative method for now.

IvanKolchanov, 2024-03-22:
> No, I have not found the solution. I am using an alternative method for now.

Sorry, I have encountered a similar problem, what is the alternative method that you are using? Thank you!

saipavan4, 2024-03-22:
Im using stereopi

saipavan4, 2024-03-22:
> > No, I have not found the solution. I am using an alternative method for now.
> 
> Sorry, I have encountered a similar problem, what is the alternative method that you are using? Thank you!

I'm using stereopi

IvanKolchanov, 2024-03-25:
@dennis-ard  Good day, I wrote an email to [support@arducam.com] with all information about the same problem that I have encountered, hope for a response about it, thank you!