ArduCAM/Arducam_tof_camera #72: Is it possible to run in headless mode on raspberry pi 4?

light-robotics, 2024-04-19:
When I'm trying to run compile.sh I'm getting the following error:
```
-- Configuring done
-- Generating done
-- Build files have been written to: /arducam_tof_camera/example/build
[ 20%] Built target test_c
[ 40%] Built target capture_raw
[ 60%] Built target preview_usb
[ 80%] Built target select_video_preview
[100%] Built target preview_depth
Unable to init server: Could not connect: Connection refused
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.1) ../modules/highgui/src/window_gtk.cpp:624: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'

Aborted
```
I'm running it in a headless mode. Don't need to display anything, just need the raw data.

Would be grateful for any help.

zoldaten, 2024-09-13:
so... just comment `cv::imshow("confidence", confidence_frame);` in Arducam_tof_camera/example/cpp examples

marcdjulien, 2026-09-16:
You can also use `ssh -Y` for X11 forwarding if you're running in headless mode.