ArduCAM/Arducam_tof_camera #73: Arducam Time of Flight camera is not recognized by (Raspberry Pi 5)

ahmetkca, 2024-05-26:
```
Machine model: Raspberry Pi 5 Model B Rev 1.0 
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
Linux version 6.6.28+rpt-rpi-2712 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22)
```

I am going through the following tutorial: https://docs.arducam.com/Raspberry-Pi-Camera/Tof-camera/Getting-Started/

I tried to make ToF camera enter into recovery mode and connect the following with dupont line:
[image]
and then power on raspberry pi 5 and then wait a little bit and then disconnect it.

However, I don't get any output when running the following command:
`dmesg | grep arducam`

<details>
  <summary>here is the full output of `dmesg`:</summary>
  
  ```
  [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.28+rpt-rpi-2712 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000015ffffff (327680 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165e0..0x000000003fd16617 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 256 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 14 pages/cpu s183976 r8192 d37208 u229376
[    0.000000] pcpu-alloc: s183976 r8192 d37208 u229376 alloc=14*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=2C:CF:67:1F:63:1F vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=272c5e5d-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=CA
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 9, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 8, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 521984
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbffc000-0x00000000ffffc000] (64MB)
[    0.000000] Memory: 7895808K/8384512K available (13376K kernel code, 2222K rwdata, 4288K rodata, 5056K init, 1213K bss, 161024K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[... 565 lines cut ...]

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

dtparam=pciex1_gen=3
#dtoverlay=ov5647
```

ahmetkca, 2024-05-26:
i even tried it on my raspberry pi 4. i got `[    8.132269] arducam-pivariety 10-000c: probe failed`

Dion4cen, 2024-05-27:
Hi,
From the config.txt file, it seems that you do not run the **Install_dependencies.sh**
https://docs.arducam.com/Raspberry-Pi-Camera/Tof-camera/Getting-Started/#step-2-change-the-directory-to-arducam_tof_camera

ahmetkca, 2024-05-27:
> Hi, From the config.txt file, it seems that you do not run the **Install_dependencies.sh** https://docs.arducam.com/Raspberry-Pi-Camera/Tof-camera/Getting-Started/#step-2-change-the-directory-to-arducam_tof_camera

Hi thanks for the reply. I tried it on my raspberry pi 5 again but this time I cloned the repo and executed the `Install_dependencies.sh` but still no luck.
here is my /boot/firmware/config.txt (this is raspberry pi 5, Debian 12 bookworm)
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=0

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

dtparam=pciex1_gen=3
#dtoverlay=ov5647
dtoverlay=arducam-pivariety,media-controller=0
```

ahmetkca, 2024-05-27:
> Hi thanks for the reply. I tried it on my raspberry pi 5 again but this time I cloned the repo and executed the `Install_dependencies.sh` but still no luck. here is my /boot/firmware/config.txt [...]

I don't know if this Time of Flight camera is entering into recovery mode successfully.

```
ahmetkca@raspberrypifive:~ $ dmesg | grep arducam
[    0.704488] platform 1f00128000.csi: Fixed dependency cycle(s) with /axi/pcie@120000/rp1/i2c@80000/arducam_pivariety@c
[    2.271947] rp1-cfe 1f00128000.csi: found subdevice /axi/pcie@120000/rp1/i2c@80000/arducam_pivariety@c
[    2.289308] arducam-pivariety 4-000c: pivariety_read: Reading register 0x103 failed
[    2.289319] arducam-pivariety 4-000c: probe failed
```

I can't update the firmware

ahmetkca, 2024-05-27:
[image]
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]

[image]
[image]

Dion4cen, 2024-05-28:
Hi,
Oh man, thank you so much for your detailed message and the pictures I need.
Let's make small changes to save this poor TOF camera. Kindly note that please connect the camera to your Pi.

- Modify **dtoverlay=arducam-pivariety,media-controller=0** to **dtoverlay=arducam-pivariety** in the config.txt file.
- Change the TOF camera to the CAM 1 port on your Pi 5's board.

ahmetkca, 2024-05-29:
> Hi, Oh man, thank you so much for your detailed message and the pictures I need. Let's make small changes to save this poor TOF camera. Kindly note that please connect the camera to your Pi.
>
> * [...]

Thanks for the reply. I have connected the ToF camera to CAM 1 port and power on the raspberry pi and then modified `dtoverlay=arducam-pivariety,media-controller=0` to `dtoverlay=arducam-pivariety` but still no luck :(

Here is the `/boot/firmware/config.txt`  file:
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=0

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

dtparam=pciex1_gen=3
#dtoverlay=ov5647
dtoverlay=arducam-pivariety
```

Here is the raspberry pi 5 and ToF connection setup:
[image]

ahmetkca, 2024-05-29:
> > Hi, Oh man, thank you so much for your detailed message and the pictures I need. Let's make small changes to save this poor TOF camera. Kindly note that please connect the camera to your Pi.
> > 
> > * Modify **dtoverlay=arducam-pivariety,media-controller=0** to **dtoverlay=arducam-pivariety** in the config.txt file.
> > * Change the TOF camera to the CAM 1 port on your Pi 5's board.
> 
> Thanks for the reply. I have connected the ToF camera to CAM 1 port and power on the raspberry pi and then modified `dtoverlay=arducam-pivariety,media-controller=0` to `dtoverlay=arducam-pivariety` but still no luck :(
> 
> Here is the `/boot/firmware/config.txt` file:
> 
> ```
> # For more options and information see
> # http://rptl.io/configtxt
> # Some settings may impact device functionality. See link above for details
> 
> # Uncomment some or all of these to enable the optional hardware interfaces
> #dtparam=i2c_arm=on
> #dtparam=i2s=on
> #dtparam=spi=on
> 
> # Enable audio (loads snd_bcm2835)
> dtparam=audio=on
> 
> # Additional overlays and parameters are documented
> # /boot/firmware/overlays/README
> 
> # Automatically load overlays for detected cameras
> camera_auto_detect=0
> 
> # Automatically load overlays for detected DSI displays
> display_auto_detect=1
> 
> # Automatically load initramfs files, if found
> auto_initramfs=1
> 
> # Enable DRM VC4 V3D driver
> dtoverlay=vc4-kms-v3d
> max_framebuffers=2
> 
> # Don't have the firmware create an initial video= setting in cmdline.txt.
> # Use the kernel's default instead.
> disable_fw_kms_setup=1
> 
> # Run in 64-bit mode
> arm_64bit=1
> 
> # Disable compensation for displays with overscan
> disable_overscan=1
> 
> # Run as fast as firmware / board allows
> arm_boost=1
> 
> [cm4]
> # Enable host mode on the 2711 built-in XHCI USB controller.
> # This line should be removed if the legacy DWC2 controller is required
> # (e.g. for USB device mode) or if USB support is not required.
> otg_mode=1
> 
> [all]
> 
> dtparam=pciex1_gen=3
> #dtoverlay=ov5647
> dtoverlay=arducam-pivariety
> ```
> 
> Here is the raspberry pi 5 and ToF connection setup: ![tof_rpi5_connection](https://private-user-images.githubusercontent.com/74574469/334658498-bf6a8726-1f11-4515-8f0e-4c19cbef609b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5NjAxMDIsIm5iZiI6MTcxNjk1OTgwMiwicGF0aCI6Ii83NDU3NDQ2OS8zMzQ2NTg0OTgtYmY2YTg3MjYtMWYxMS00NTE1LThmMGUtNGMxOWNiZWY2MDliLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI5VDA1MTY0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU4ODI1ZDI4ZGE4YzRkYWJhMzE3MTc0Mjc2NTZmMzdmNzY1OTg5NTg1MTJjOTg5MGExYmRkN2QzZmNmOWEwNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.HfqmZDSOA8wmpX7n-3p_fWAS47OzBptAvnmZNzHFItA)

@Dion4cen In the dmesg output I still get `[    2.281430] arducam-pivariety 4-000c: probe failed`

Dion4cen, 2024-05-29:
Hi @ahmetkca 
The cable to the camera might be flipped over to the wrong side. Try to disconnect the cable, flip it over, and reconnect
[image]

ahmetkca, 2024-05-29:
Ok even though I got `probe failed` in `dmesg` output but according to https://docs.arducam.com/Raspberry-Pi-Camera/Tof-camera/Getting-Started/#update-firmware you should see the following output if you successfully make the ToF camera go into recovery mode:
`$ dmesg | grep arducam`
[image]

I tried to update the firmware anyways and it was successful this time even though dmesg says otherwise:
```
ahmetkca@raspberrypifive:~/Downloads/B0410_Tof_Firmware/aarch64/b0410_tof_firmware $ sudo dmesg | grep arducam-
[    2.281430] arducam-pivariety 4-000c: probe failed
ahmetkca@raspberrypifive:~/Downloads/B0410_Tof_Firmware/aarch64/b0410_tof_firmware $ sudo ./firmware_update -i 4 -m b0410     _tof_firmware
[INFO] Firmware Version: 0xFFFFFFFE
[INFO] Sensor ID: 0xFFFFFFFE
[INFO] ==================First stage==================
[INFO] Write update flag 2.
[INFO] Write checksum.
[INFO] Write update flag.
[INFO] Sending... |======================================| ETA: 0h00m05s
[INFO] All data has been sent.
[INFO] Updating...
[SUCCESS] Firmware update successful.
[INFO] Firmware Version: 0x10002
[INFO] Sensor ID: 0x2311
[SUCCESS] ==================First stage success==================
[INFO] ==================Second stage==================
[INFO] Write update flag.
[INFO] Write checksum.
[INFO] Sending... |======================================| ETA: 0h00m02s
[INFO] All data has been sent.
[INFO] Updating...
[SUCCESS] Firmware update successful.
[SUCCESS] ==================Second stage success==================
[SUCCESS] Firmware update successful, please reboot.
ahmetkca@raspberrypifive:~/Downloads/B0410_Tof_Firmware/aarch64/b0410_tof_firmware $ sudo reboot -p --reboot

Broadcast message from root@raspberrypifive on pts/1 (Wed 2024-05-29 01:30:22 EDT):

The system will reboot now!

ahmetkca@raspberrypifive:~/Downloads/B0410_Tof_Firmware/aarch64/b0410_tof_firmware $ Connection to raspberrypifive.local closed by remote host.
Connection to raspberrypifive.local closed.
❯ ssh -a -t -o Compression=yes -i ~/.ssh/id_rsa ahmetkca@raspberrypifive.local
Linux raspberrypifive 6.6.28+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed May 29 01:30:43 2024
ahmetkca@raspberrypifive:~ $ sudo dmesg | grep arducam
[    0.704615] platform 1f00128000.csi: Fixed dependency cycle(s) with /axi/pcie@120000/rp1/i2c@80000/arducam_pivariety@c
[    2.221115] arducam-pivariety 4-000c: firmware version: 0x10002
[    2.232612] rp1-cfe 1f00128000.csi: found subdevice /axi/pcie@120000/rp1/i2c@80000/arducam_pivariety@c
[    2.422407] rp1-cfe 1f00128000.csi: Using sensor arducam-pivariety 4-000c for capture
```

After successfully updating the firmware, I continued from pulling the `Arducam_tof_camera` repo https://docs.arducam.com/Raspberry-Pi-Camera/Tof-camera/Getting-Started/#step-3-installation-driver-dependencies-sdk-opencv
```
ahmetkca@raspberrypifive:~ $ sudo dmesg  | grep arducam-
[    2.232867] arducam-pivariety 4-000c: firmware version: 0x10002
[    2.468185] rp1-cfe 1f00128000.csi: Using sensor arducam-pivariety 4-000c for capture
```

Success!
[image]

closed 2024-05-29 by ahmetkca