ArduCAM/Arducam_tof_camera #74: Raspberry Pi Zero W support, suspected problem with armv6 ISA

Alvarezzccc, 2024-05-29:
Hello,

I am working on a project for low power consumption using a Raspberry Pi Zero W with ARMv6 architecture. I suspect that some error occurred after I installed the newest versions of the "arducam-config-parser-dev", "arducam-usb-sdk-dev", and "arducam-tof-sdk-dev" packages. I am always getting an "Illegal Instruction" error message when executing my Arducam ToF C++ program. When debugging with the gdb tool, I saw that the CPU halts at the instruction "cbnz r2, 0xb5f609ec" (compare and branch if non-zero), which I guess is more complex than what this ARMv6 ISA is supposed to handle. This instruction is in the /lib/libarducam_config_parser.so library, which comes from one of the previously mentioned packages. These versions are respectively:

`
arducam-tof-sdk-dev:
  Candidate: 0.1.3

arducam-config-parser-dev:
  Candidate: 0.1.2

arducam-usb-sdk-dev:
  Candidate: 2.0.1
`

Do you know a way to be able to access to the latest versions of these libraries? I was not able to access previous ones in the link proportionated by apt-cache. 

[image]

langet0695, 2025-05-15:
Hi @Alvarezzccc - Did you ever figure this out? I've the Raspberry Pi Zero W with ARMv6 architecture and seem to be having the same issue you were.

langet0695, 2025-06-05:
Legendary!! Thank you for sharing. That worked.