ArduCAM/Arducam_tof_camera #85: Error "libpng warning: iCCP: known incorrect sRGB profile"

drwillis, 2024-08-12:
I am getting this error message when running Arducam_tof_camera/build/example/cpp/example and Arducam_tof_camera/build/example/cpp/preview_depth targets.

Running the code on an Arducam TOF camera with newly flashed firmware on a PI 5 with bookworm debian 64bit arm OS with a 6.6.31-rpt-rpi-2712 kernel. The camera was working before a recent update. Versions follow:

arducam-tof-skd-dev 0.1.9
libpng-dev 1.6.39-2

The error is shown below:
pidar@pidar5:~ $ Arducam_tof_camera/Release/example/cpp/preview_depth 
open camera with (240x180)
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile

drwillis, 2024-08-13:
This problem was resolved after an update and a subsequent reboot. Apologies for the spurious report.

batuhanuzaldimvc, 2024-08-14:
Having the same issue. It has just appeared with the very recent installations. Our older (a couple of weeks ago) ToF installations were fine. We could not solve it by just updating and rebooting the system as mentioned above.

OS: RPi Bookworm 64-bit
Host Devices: RPi 4 & Compute Module 4
B0410 Firmware: The old/original one and the new one. We updated one of our ToF just to see if the older firmware was the issue, it wasn't. apparently.