ArduCAM/Arducam_tof_camera #93: Reported depth measurement: point to point or point to virtual plane?

diplodocuslongus, 2024-12-13:
How does the Arducam ToF camera reports depths?
1) from the center of the lens to each location of the target ("range" in the drawing below)?
2) Or each points location in the target is projected on a virtual plane passing through the sensor plane ("depth" in the drawing)?

[image]

From our measurements it seems that depths are reported according to 1), that is, there is no correction for the angle between the optical axis of the lens and the ray between the center of the lens to a target object.

diplodocuslongus, 2024-12-26:
The [answer](https://forum.arducam.com/t/measurement-from-the-tof-camera/7729) provided by Arducam on their forum is that the distance reported by the ToF sensor is the depth (Z) above, not the range (R). 

However when we measure the depth from the ToF to a straight object over the full length of the object, the distance isn't constant, it appears curved, as seen below. Note that the distance at each points is averaged over 10 measurements. 

The setup is shown below.
[image]
[image]

[image]

We probably can ignore the extreme points as they may be noisier (near the edge) but the points that remain show the distance is not constant. Also the ground truth (i.e. accuracy of the measurement) is off by almost 10% (ground truth along the optical axis of the camera to the object is ~57cm, measured is ~66cm.

jeremfrs, 2025-01-17:
Hello, I was wondering the same thing as you because i noticed some rounding at the edges on my depth image, so I ask myself in the depth image was a depth map (point to plane distance) or a zbuffer (point to point distance). However, from further testing this "rounding artefact" disappeared on other capture for me, i can't explain why. If you are not convinced you can easily convert a zbuffer to a depth map (and vice versa) by dividing each pixel value by the following coefficient
$c(x,y) = \sqrt{1 + (x - c_x)^2/f_x^2 + (y-c_y)^2/f_y^2}$
However as you will notice, this does not improve the "rounding". Also, in the preview_pointcloud.cpp in this repo, you can notice they use the open3d::geometry::PointCloud::CreateFromDepthImage function that takes a depth map as input.