ArduCAM/Arducam_tof_camera #101: Install_dependencies_jetson.sh Error

DimIsaev, 2025-01-23:
./Install_dependencies_jetson.sh

**Error**

```bash
Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  4.9.337-tegra-32.7.6-20241104234540
Jetson type:  NVIDIA Jetson Nano Developer Kit
```

env

[image]

DimIsaev, 2025-02-27:
Is there anyone?

DimIsaev, 2025-03-23:
Answr ?

clankill3r, 2025-12-09:
@DimIsaev libopen3d-dev is not in pip as well. And I also I couldn't get the C and CPP examples to run.
People should learn to bundle dependencies with their project. I'm still amazed how easy is it is to get quake 3 running from source but it not using pip probably helps a lot :)